eliteSchwein/mooncord #2: Notification?

gcormier, 2021-03-16:
Any thoughts to adding notifications in the channel when a print completes?

eliteSchwein, 2021-03-27:
hey there, it should work :) sorry, i was kinda inactive 
[image] if it doesnt show up, try to post your output from systemctl status mooncord

gcormier, 2021-03-27:
Nope, no print done status

```
● MoonCord.service - MoonCord
   Loaded: loaded (/etc/systemd/system/MoonCord.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2021-03-24 20:23:19 EDT; 2 days ago
 Main PID: 651 (npm)
    Tasks: 23 (limit: 4915)
   CGroup: /system.slice/MoonCord.service
           ├─651 npm
           ├─697 sh -c node mooncord.js
           └─698 node mooncord.js

Mar 26 11:55:44 mainsailos npm[651]: Printer Status: printing
Mar 26 12:00:01 mainsailos npm[651]: Printer Status: printing
Mar 26 12:04:22 mainsailos npm[651]: Printer Status: printing
Mar 26 12:08:39 mainsailos npm[651]: Printer Status: printing
Mar 26 12:12:56 mainsailos npm[651]: Printer Status: printing
Mar 26 12:17:13 mainsailos npm[651]: Printer Status: printing
Mar 26 12:21:47 mainsailos npm[651]: Printer Status: done
Mar 26 12:21:48 mainsailos npm[651]: Printer Status: ready
Mar 26 12:29:51 mainsailos npm[651]: Printer Status: disconnected
Mar 26 12:29:55 mainsailos npm[651]: Printer Status: ready
```
Do I need any sort of specific `M117` status gcode when finished a print for mooncord to detect that it's done? Bot is working OK otherwise, responds to my commands.

eliteSchwein, 2021-03-27:
You see the message Printer Status: done? When this comes the done message appears, do you have configured a broadcast channel? (Prefix:addchannel)?

gcormier, 2021-03-28:
Ah excellent :) Sorry I didn't realize you had to add a broadcast channel. Working now!

closed 2021-03-28 by gcormier

eliteSchwein, 2021-04-08:
@gcormier i added in 0.0.2 even now dm notifications :)