eliteSchwein/mooncord #7: Feature request / guidance requested

bdkts, 2021-04-27:
I'd love to have a mechanism to allow klipper to send a custom message directly through the bot. Here's an example of a macro using a telegram script to do this: https://github.com/th3fallen/voronConfig/blob/master/macros.cfg#L335

I'd be glad to implement this myself but though I'd check with you to get some guidance in case you had some pre-formed thoughts on the subject. 

Any ideas/opinions?

bdkts, 2021-04-27:
btw, one thought I had was just to use a simple tmp file as a message bus... make a script to dump a pre-formatted json object in the file and have the bot watch for content in this file. It's hacky, but simple to accomplish.

eliteSchwein, 2021-04-27:
i put it on my 0.0.3 Project List, i think i will make a klipper plugin for that task

eliteSchwein, 2021-05-14:
https://github.com/eliteSchwein/mooncord/wiki/GCode-Commands wiki entry is already there

eliteSchwein, 2021-05-27:
[image]
86ac2371d414a384ce1348abae03f080eb846080
81a4333b93521d299d9b4bcdb0708ae0f0821a1f

closed 2021-07-15 by eliteSchwein