eliteSchwein/mooncord #17: /editchannel - This Command is only aviable on a Guild

stefanpausch, 2021-05-01:
Hello,

If i use the **/editchannel** command the Bot replies with "This Command is only aviable on a Guild"

If i use the **/editchannel** -> _TAB_ -> _Channel highlight with cursor keys_ -> _TAB_ -> _ENTER_ i get "A channel id specified is invalid"

If i use **/editchannel _MY_CHANNEL_ID_I_WANT_TO_USE_** Bot replies with "This Command is only aviable on a Guild"

Is this a bug, or am i using the command not correctly?

What does the  "This Command is only aviable on a Guild" output mean?

eliteSchwein, 2021-05-01:
I will test it

eliteSchwein, 2021-05-01:
i tested it, it works fine for me

[image]

stefanpausch, 2021-05-01:
Thanks alot for testing. Tested it again and sadly still not working

- Configuration is correct (API IDs, etc)
- Bot is connected
- All other commands (status, info, etc) are working fine (getting output)
- Bot is showing current status in its status line
- Only editchannel is not working

- The bot only responds to PM commands (i think thats expected behaviour, is it?)
- /admin user TAB only sees the bot name, not other users on the server (maybe thats a hint for the issue?

Is there anything else i need to configure, besides what is written in die README?

Is there anything i can test?

eliteSchwein, 2021-05-01:
whats the output of systemctl status MoonCord after executing the command?

stefanpausch, 2021-05-01:
Output of **sudo systemctl status MoonCord**

```bash

● MoonCord.service - MoonCord
   Loaded: loaded (/etc/systemd/system/MoonCord.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2021-05-01 15:28:39 BST; 53min ago
 Main PID: 673 (npm)
    Tasks: 23 (limit: 4915)
   CGroup: /system.slice/MoonCord.service
           ├─673 npm
           ├─724 sh -c node mooncord.js
           └─725 node mooncord.js

May 01 15:28:45 mainsailos npm[673]:   Moonraker Client Connected
May 01 15:28:45 mainsailos npm[673]:   Sent initial Moonraker commands
May 01 15:28:45 mainsailos npm[673]:   Initial Automatic Moonraker commands
May 01 15:28:45 mainsailos npm[673]: ℹ Printer Status: ready
May 01 15:28:46 mainsailos npm[673]: ℹ There are some Updates!
May 01 15:31:10 mainsailos npm[673]: ℹ Printer Status: disconnected
May 01 15:31:15 mainsailos npm[673]: ℹ Printer Status: ready
May 01 15:51:44 mainsailos npm[673]: ℹ Printer Status: start
May 01 15:51:44 mainsailos npm[673]: ℹ Printer Status: printing
May 01 16:14:33 mainsailos npm[673]: ℹ Printer Status: printing`
```
> wait do you execute the admin and edit command on a Discord Server (Guild)?

- I am connected (as owner) to a discord server, using the official discord windows client
- the bot is connected to the server and shows up as user (added with https://discord.com/oauth2/authorize?scope=bot&permissions=0&client_id=YOUR_CLIENT_ID_HERE) 
- The bot commands only work if i direct message the bot 
- The bot commands DON'T show up in a channel (correct?)
- Is there any other way to use the commands besides DM and within a channel?

eliteSchwein, 2021-05-01:
okay i see the problem, you should use the invite link from the bot... (it appear on startup from the bot)

eliteSchwein, 2021-05-01:
https://discord.com/oauth2/authorize?client_id=CLIENT_ID&permissions=3422944320&scope=bot%20applications.commands

this is the proper invite link

stefanpausch, 2021-05-01:
Perfect! It is working now :) - Thanks alot for the help.

Would you mind updating your README with the information?

eliteSchwein, 2021-05-01:
it should stand there

eliteSchwein, 2021-05-01:
the URL is visible on startup, but yeah i will mention that in the Readme

stefanpausch, 2021-05-03:
I reopen the issue. My Bot is working fine. I am the guild owner, but: 

A friend of mine added his own app/own bot to the server and everything is working fine, but the /editchannel command. If i use the command the bot outputs "An Error occured!"

The bot was added withe the correct url by himself and my friend has the rights to add bots to the server.

Is there any way to debug this? (i will send my friend the url to this issue and he can add more information to it later.

Thanks alot!

eliteSchwein, 2021-05-03:
Can you send the log pls?

systemctl status MoonCord

Is the command

venom9982, 2021-05-03:
Hello, systemctl status MoonCord does not show anything about the problem:

systemctl status MoonCord
pi@mainsailos:~ $ sudo systemctl status MoonCord
● MoonCord.service - MoonCord
   Loaded: loaded (/etc/systemd/system/MoonCord.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-05-03 17:10:49 BST; 32s ago
 Main PID: 19593 (npm)
    Tasks: 23 (limit: 4915)
   CGroup: /system.slice/MoonCord.service
           ├─19593 npm
           ├─19611 sh -c node mooncord.js
           └─19612 node mooncord.js

May 03 17:10:53 mainsailos npm[19593]:   | |\/| / _ \/ _ \ ' \| '_/ _` | / / -_) '_|
May 03 17:10:53 mainsailos npm[19593]:   |_|  |_\___/\___/_||_|_| \__,_|_\_\___|_|
May 03 17:10:53 mainsailos npm[19593]:
May 03 17:10:53 mainsailos npm[19593]:   Connect to Moonraker
May 03 17:10:53 mainsailos npm[19593]:   Enable Moonraker Events
May 03 17:10:53 mainsailos npm[19593]:   Moonraker Client Connected
May 03 17:10:53 mainsailos npm[19593]:   Sent initial Moonraker commands
May 03 17:10:53 mainsailos npm[19593]:   Initial Automatic Moonraker commands
May 03 17:10:53 mainsailos npm[19593]: ℹ Printer Status: shutdown
May 03 17:10:53 mainsailos npm[19593]: ℹ There are some Updates!

[image]

The link https://discord.com/oauth2/authorize?client_id=CLIENT_ID&permissions=3422944320&scope=bot%20applications.commands is the one I already used.

Any idea? :)

I'll try to delete the Bot and reinvite

venom9982, 2021-05-03:
Ok, now I get this:

MoonCord.service - MoonCord
   Loaded: loaded (/etc/systemd/system/MoonCord.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-05-03 17:22:22 BST; 2min 37s ago
 Main PID: 19874 (npm)
    Tasks: 23 (limit: 4915)
   CGroup: /system.slice/MoonCord.service
           ├─19874 npm
           ├─19886 sh -c node mooncord.js
           └─19887 node mooncord.js

May 03 17:22:26 mainsailos npm[19874]: ℹ There are some Updates!
May 03 17:22:26 mainsailos npm[19874]: (node:19887) UnhandledPromiseRejectionWarning: DiscordAPIError: Cannot send messages to this user
May 03 17:22:26 mainsailos npm[19874]:     at RequestHandler.execute (/home/pi/mooncord/node_modules/discord.js/src/rest/RequestHandler.js:154:13)
May 03 17:22:26 mainsailos npm[19874]:     at processTicksAndRejections (internal/process/task_queues.js:86:5)
May 03 17:22:26 mainsailos npm[19874]: (node:19887) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handle
May 03 17:22:26 mainsailos npm[19874]: (node:19887) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
May 03 17:22:27 mainsailos npm[19874]: (node:19887) UnhandledPromiseRejectionWarning: DiscordAPIError: Cannot send messages to this user
May 03 17:22:27 mainsailos npm[19874]:     at RequestHandler.execute (/home/pi/mooncord/node_modules/discord.js/src/rest/RequestHandler.js:154:13)
May 03 17:22:27 mainsailos npm[19874]:     at processTicksAndRejections (internal/process/task_queues.js:86:5)
May 03 17:22:27 mainsailos npm[19874]: (node:19887) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handle

eliteSchwein, 2021-05-03:
1. please dont post a invite Link with your Bots ID, everyone can use that URL....
2. you can only use chat Channels, did you try a announcment channel?
3. the second message, the Bot cant DM someone and that causes some internal issues... i will fix that, but please make a new Issue for that :)

venom9982, 2021-05-03:
1. No, I only tried to use an Chat Channel

stefanpausch, 2021-05-03:
I will close the issue again :) Thanks for the help!

closed 2021-05-03 by stefanpausch