eliteSchwein/mooncord #21: Automatic channel status - percentage based on time

stefanpausch, 2021-05-03:
I would love to see a more detailed options for the automatic channel status posting.  

Currenly it is only based on percentage of a print. If i print only takes a couple of minutes to print the status is spammed way to often.

Not sure what is the best fit here but i would love seeing something like "post every 10%, but only if last message was 30min ago, or post if last message was 2 hours ago if if 10% haven't reached yet"

Thanks alot!

eliteSchwein, 2021-05-03:
You mean minimum Time between Messages?

eliteSchwein, 2021-05-04:
Will come in 0.0.3

eliteSchwein, 2021-05-14:
https://github.com/eliteSchwein/mooncord/wiki/Status-Configuration wiki entry is now present

eliteSchwein, 2021-05-20:
implemented in the dev tree

6cca8c5d484b5c5c9ebfe59d5e77d176c33330e4
c50d18a6b7e438663f14fdb463ac4d774c2d3d91

closed 2021-07-15 by eliteSchwein