eliteSchwein/mooncord #35: Printer Status Error With Multi Printer on 1 PI

screech24, 2021-05-16:
I have Tried to install MoonCord with a Multiple Klipper Instance on 1 PI setup where each printer is served on a different moonraker Port EX. 7126 or 7125

here is the error im getting using http://127.0.0.1:7126 as the moonrakerurl in the config,json
[image]
I am also unable to Change Permissions in the discord Server but im not sure this is a related issue. 
[image]

eliteSchwein, 2021-05-16:
From the discord converation:

Master ID is your discord id and application id is the id from your application...

Also you need to change the port on the websocket (ws://127.0.0.1:7126)

eliteSchwein, 2021-05-16:
Solved via discord

closed 2021-05-16 by eliteSchwein