eliteSchwein/mooncord #62: Feature Request: Smart Outlet On/Off

sashi, 2021-06-01:
Use case: To have lights controlled by smart outlet to be turned on and off with a status update (/status or automatic interval)

Telegram Klipper bot has this
Example
`#Led on link for picture
led_on=""
#Led on wait time before picture is taken (in seconds) 
led_on_delay="0"
#Led off link for picture
led_off=""
#Led off wait time after picture is taken (in seconds)
led_off_delay="0"`

It is being controlled via HTTP request with url "lighton" and "lightoff". Currently using it manually via GCODE in klipper as below

`[gcode_shell_command lighton]
command: sh /home/pi/light/lighton.sh
timeout: 30.
verbose: True

[gcode_macro LIGHTON]
gcode:
    RUN_SHELL_COMMAND CMD=lighton

[gcode_shell_command lightoff]
command: sh /home/pi/light/lightoff.sh
timeout: 30.
verbose: True

[gcode_macro LIGHTOFF]
gcode:
    RUN_SHELL_COMMAND CMD=lightoff`

eliteSchwein, 2021-06-01:
Okay but timing would be a issue i think... I will implement it tomorrow in the dev tree

eliteSchwein, 2021-06-11:
implemented in the dev tree f82201c9011322da93fc5652b4d6cb15ca94bd2a

closed 2021-07-15 by eliteSchwein