eliteSchwein/mooncord #76: Print time shows the slicer estimated time not the actual printed time

jarheck, 2021-07-13:
When getting the print status it shows the "Print Time" as the slicer estimated time instead of the actual print time. I am not sure if this is expected or a bug? If it is a bug maybe this is a feature request to show the actual time elapsed for the print.

eliteSchwein, 2021-07-13:
currently i use the slicer estimated time, but i will replace it with the klipper estimated time in 0.0.3

jarheck, 2021-07-13:
Would it be possible to get the actual elapsed time for the print in the Print Done message?

eliteSchwein, 2021-07-13:
Sure!

eliteSchwein, 2021-07-13:
fixed in 79ee251001cd10320eca5b6262fa204c55366a8e

eliteSchwein, 2021-07-13:
Reopen because i made it even worser now... I will tomorrow update the methode for the time stuff

eliteSchwein, 2021-07-15:
Finally solved with a heck lot other buggs

closed 2021-07-15 by eliteSchwein