eliteSchwein/mooncord #105: Webcam Issue

vertigo235, 2021-08-15:
Getting this error 

 Webcam Util: Error: unknown JPEG marker ffd6

Is it possible that my camera settings could be the issue?

camera_usb_options="-r 1280x720 -f 30"

Works fine in Fluidd

eliteSchwein, 2021-08-15:
its solved in 0.0.4, it was a issue with the jimp library

vertigo235, 2021-08-15:
Nice, what branch do I switch to for the bleeding edge? dev?

eliteSchwein, 2021-08-15:
dev but you need node v16 and i need to rework all commands to work with discordjs v13

vertigo235, 2021-08-15:
Is there a workaround for now? I can't seem to find any specific commit in your dev branch that addresses the issue?  I was going to cherry pick it.

eliteSchwein, 2021-08-15:
I replaced jimp with sharp, the class is in utils named webcamUtil

vertigo235, 2021-08-15:
Can you share the commit?

vertigo235, 2021-08-15:
ahh this one I presume https://github.com/eliteSchwein/mooncord/commit/0b4c08a322e8d0c88cc2b0af5f63656f9d12b20d#diff-6553ba0bec03413bae0fbde8ddfeb03aed61eacd3459cea7ff2c2b3b9a57c5be

vertigo235, 2021-08-15:
errr this one too https://github.com/eliteSchwein/mooncord/commit/221399957e90aaa337ebb64282d8ac67c870fc91#diff-6553ba0bec03413bae0fbde8ddfeb03aed61eacd3459cea7ff2c2b3b9a57c5be

eliteSchwein, 2021-08-15:
Yep these commits

vertigo235, 2021-08-15:
Well I gave an honest effort but I'm having trouble getting the sharp dependency installed, I even tried to run the dev branch but had a similar issue. I'll just wait I guess, thanks for your work on this great tool!

eliteSchwein, 2021-09-02:
*hopefully* resolved in 0.0.4

closed 2021-09-02 by eliteSchwein