eliteSchwein/mooncord #119: Update complicated from 0.3.1 to 0.4.0

top-gun, 2021-09-04:
I installed MoonCord 0.3.1 on my Fluiddpi which uses the most current Klipper, Moonraker and OS files.

As I just got the update notification for 0.4.0, I updated via the Fluidd-UI. During the update, I noticed some warnings about cleaning up folders in NPM.

After the update, MoonCord no longer worked, I got only "this interaction has failed" to any command.

Steps I took to resolve the issue (successfully):

1.) sudo chmod +X +R and +W for the folder "scripts" (I could only see the filenames, not even read permissions)
2.) sudo ./migrateNode.sh (this was probably unnecessary, I got a warning from the updater)
3.) ./migrateNode.sh (as Pi, no su rights)

After a restart and a few minutes, MoonCord is working again.

eliteSchwein, 2021-09-06:
I stated already on a newer issue, i am aware about the bugs in 0.0.4, before i released it, it worked great... I tested it very deeply till there where no issues anymore, but every setup is different it seems like, i try to fix the issues asap

top-gun, 2021-09-06:
Thank you for your enormous effort. The software is very handy and I prefer it over VPN when I'm not at home. :)

eliteSchwein, 2022-01-07:
resolved in 0.0.5

closed 2022-01-07 by eliteSchwein