eliteSchwein/mooncord #120: 0.4.12 crashes with some commands like /listfiles

top-gun, 2021-09-06:
0.0.4-12 crashes when querying some data

This is repeatable. I am using a fresh installation after some issues with updates from 0.3.1. The system was cleaned with the two commands mentioned in the Wiki (fwiw, should be "rm -rf..." instead of "rm rf...")

I added app and bot keys, also the user-id.

In general, I can do simple tasks like /info

When I do stuff that involves Moonraker, I get an error about failed interactions, and MoonCord crashes on the pi.

[image]

pi@vcore:~ $ sudo systemctl status MoonCord
● MoonCord.service - MoonCord
   Loaded: loaded (/etc/systemd/system/MoonCord.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-09-06 16:34:35 CEST; 10s ago
  Process: 5366 ExecStart=/usr/local/bin/npm run start /home/pi/klipper_config/ (code=exited, status=1/FAILURE)
 Main PID: 5366 (code=exited, status=1/FAILURE)

Sep 06 16:34:21 vcore npm[5366]: ℹ Printer Status: ready
Sep 06 16:34:35 vcore npm[5366]: ✖ List Files Command: TypeError: Cannot read property 'id' of null
Sep 06 16:34:35 vcore npm[5366]: /home/pi/mooncord/discord/commands/listFiles.js:58
Sep 06 16:34:35 vcore npm[5366]:         connection.removeListener('message', handler)
Sep 06 16:34:35 vcore npm[5366]:                    ^
Sep 06 16:34:35 vcore npm[5366]: TypeError: Cannot read property 'removeListener' of undefined
Sep 06 16:34:35 vcore npm[5366]:     at Object.module.exports.reply (/home/pi/mooncord/discord/commands/listFiles.js:58:20)
Sep 06 16:34:35 vcore npm[5366]:     at processTicksAndRejections (node:internal/process/task_queues:96:5)
Sep 06 16:34:35 vcore systemd[1]: MoonCord.service: Main process exited, code=exited, status=1/FAILURE
Sep 06 16:34:35 vcore systemd[1]: MoonCord.service: Failed with result 'exit-code'.
pi@vcore:~ $

top-gun, 2021-09-06:
Seems like some argument was written wrong.

eliteSchwein, 2021-09-06:
I am aware about the current bug Situation... I am not sure why the bugs appear.. i tested everything and it worked great....

top-gun, 2021-09-06:
I think this was my fault, as I misspelled some entries. Fwiw, I feel some sentences as explanation "what goes where" and a fully edited config (with 1111, 2222,3333 of course) is easier to follow than the animations.

eliteSchwein, 2021-09-09:
okay i will make a example config soon

chrisepp, 2021-12-07:
I have the same issue. Everything was working, new install today, and after I performed /listfiles it is crashed. same error.

eliteSchwein, 2022-01-07:
resolved in 0.0.5

closed 2022-01-07 by eliteSchwein