eliteSchwein/mooncord #147: Make logging to file optional

ressu, 2021-10-31:
It should be possible to both define the log file location and to disable the log file completely.

While it's beneficial in some cases to have a log file, in other cases it's not as useful. For example, when running in a Kubernetes container, the standard way of logging information is via stdout and stderr. If I want to run the container with a readonly root, I will need to separately prepare for log. Even though there are other uses for the temp directory too, this is one of the obvious things that should be configurable.

eliteSchwein, 2021-11-02:
about the log path, sure i can add it but disabling the log doesnt make any sense, mooncord is still in Alpha and when someone finds a bug and cant provide a log its gonna get a hard time to find out what causes the issue

ressu, 2021-11-02:
Hah, I was about to discuss the log file requirement, but you happened to edit the comment while I was writing this.

I would prefer to have the temp directory as a configurable value too. Preferably following the [XDG base directory standard](https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html) specification, but even having an argument to change it is a good start.

eliteSchwein, 2021-11-02:
configurable temp dir aswell? okay i will add it

eliteSchwein, 2021-11-04:
implemented in 5d13788d396fd0bea0434d54c31173d0ce464546

closed 2021-11-04 by eliteSchwein