eliteSchwein/mooncord #152: When printing a gcode file in a folder the gcode preview image is not shown in discord.

samwiseg0, 2021-11-09:
#### Issue description
When printing a gcode file in a folder the gcode preview image is not shown.

#### Steps to reproduce the issue
1.  Upload a gcode file to klipper/moonraker.
2. Organize it into a folder
3. Send the file to print
4. Observe that the notification is using the generic image for the file and not the preview image

#### What's the expected result?
When printing from a subfolder the preview image should be present in the notification on discord. 

#### What's the actual result?
The preview image is the default and not the one contained in the gcode file. 

#### Additional details / screenshot
In a subfolder.
[image]

Not in a subfolder.
[image]

 
#### Version Info
`MoonCord v0.0.4-14`

eliteSchwein, 2021-11-09:
i will fix that with 0.0.5, since its only aesthetics there will be no hotfix for that in 0.0.4

eliteSchwein, 2022-01-07:
resolved in 0.0.5

samwiseg0, 2022-01-07:
This still appears to be an issue
[image]
?

samwiseg0, 2022-01-08:
I confirmed this is an issue when retrieving the file. It is missing the sub folder path. 

```
[error] [2022-01-08T22:23:35.166Z] Thumbnail Error:
[error] [2022-01-08T22:23:35.167Z] Url: http://127.0.0.1/server/files/gcodes/.thumbs/SB_CW2_beta3_0.2mm_T240_B110_R0.85mm_RS50_EM0.9304_eSUN_Black_ABS+_Voron_v2_11h25m.arcw-400x300.png
[error] [2022-01-08T22:23:35.167Z] Error: Error: Request failed with status code 404
```

It should be:
```
http://127.0.0.1/server/files/gcodes/SUBFOLDERHERE/.thumbs/SB_CW2_beta3_0.2mm_T240_B110_R0.85mm_RS50_EM0.9304_eSUN_Black_ABS+_Voron_v2_11h25m.arcw-400x300.png
```

eliteSchwein, 2022-01-10:
[image]
i tried to reproduce it, it worked for me just fine

eliteSchwein, 2022-01-10:
fixed with febcff0f18b75d39c124cc4705fc8da7b95e646b

closed 2022-01-10 by eliteSchwein