eliteSchwein/mooncord #195: Pause/resume not functioning properly with v0.0.5

samwiseg0, 2022-01-08:
WIth version 0.0.5 pause and resume do not functioning properly. When the print is paused and then resumed it just states print started and the print job status is not updated there after. 

[image]

Logs
```
```

samwiseg0, 2022-01-08:
As a note this error seems to always be present in the logs.  I have also configured periodic status updates and they are not firing. I am not sure if they are related or If I should open another issue just for that

```
[error] [2022-01-08T22:18:49.037Z] Error: Timed out after waiting for 20000 ms
            Error: Timed out after waiting for 20000 ms
    at /home/pi/mooncord/dist/index.js:7:849
    at async StatusHelper.update (/home/pi/mooncord/dist/index.js:49527:9)
 ```

eliteSchwein, 2022-01-10:
i think i know where `Error: Timed out after waiting for 20000 ms` gets triggert

eliteSchwein, 2022-01-10:
resume message is fixed with 355b855cb05ea368566cb802bdafef6e3cbd2b90
on the scheduler part: i will test it first, i think i forgot to implement it

eliteSchwein, 2022-01-10:
scheduler fixed with 30051129fc5f0b34d775c9ddc492be9d20f42e3b

closed 2022-01-10 by eliteSchwein