eliteSchwein/mooncord #202: Installation fails on ARM64

cowboy3d, 2022-01-10:
When installing on ARM64 (raspberry pi 4 with raspian 64 bit) the script "scripts/install.sh" fails when downloading and installing the node libraries:
  if ! command -v node -v >/dev/null 2>&1
    then
        status_msg "Download Node 16.13.1"
        wget https://nodejs.org/download/release/v16.13.1/node-v16.13.1-linux-armv7l.tar.gz

Here a check is needed if we are on ARM64 and the download needs to be
        wget https://nodejs.org/download/release/v16.13.1/node-v16.13.1-linux-arm64.tar.gz

eliteSchwein, 2022-01-10:
i am thinking about implementing this:

```
curl -sL https://deb.nodesource.com/setup_16.x | sudo -E bash -
sudo apt-get install -y nodejs (auto corrected depending on the plattform)```

eliteSchwein, 2022-01-10:
implemented with 3c071b2c36cdc1dc1b57ca0b6f4a70bc62fa61fa and e9419d581c2fb02c349706a7d8fbe631db96a708

closed 2022-01-10 by eliteSchwein