eliteSchwein/mooncord #254: [FR] Dynamically set the color of a broadcast message

samwiseg0, 2022-02-09:
The ability to send a color code with a broadcast message to easily differentiate the type of message received. 

```
RESPOND PREFIX=mooncord.broadcast MSG="Runout detected!" COLOR="#cc0000"

RESPOND PREFIX=mooncord.broadcast MSG="We are ready to print!" COLOR="#4cf502"
```

eliteSchwein, 2022-03-14:
the argument COLOR=" is not a valid syntax for klipper, sadly but my solution would be

`RESPOND PREFIX=mooncord.broadcast MSG="Runout detected!COLOR:#cc0000" `

eliteSchwein, 2022-03-14:
okay updates COLOR is fine, the # is a issue

samwiseg0, 2022-03-14:
> okay updates COLOR is fine, the # is a issue, but it doesnt get send to mooncord...
> 
> `RESPOND PREFIX=mooncord.broadcast MSG="Runout detected!COLOR:cc0000"`
> 
> is the only solution

That works for me. I think this is the best way to go. This would also allow some other customization in the future if you chose to implement any other options. 

Thank you for taking time time to implement this