eliteSchwein/mooncord #285: Mooncord crashing when canceling a Print

Agent772, 2022-04-24:
Hi,

Mooncord crashes when canceling a print via Mooncord or KlipperScreen.

Here is the Log:

[info] [2022-04-23T07:44:19.137Z] Agent772#3294 pressed button: printjob_refresh
[info] [2022-04-23T07:44:38.632Z] Agent772#3294 pressed button: printjob_cancel
[info] [2022-04-23T07:44:39.262Z] executing macro: CANCEL_PRINT
[info] [2022-04-23T07:44:39.684Z] klipper status changed to cancelled...

[error] [2022-04-23T07:44:39.756Z] TypeError: Cannot read properties of undefined (reading 'embed_id')
            TypeError: Cannot read properties of undefined (reading 'embed_id')
    at StatusHelper.update (/home/pi/mooncord/dist/index.js:51408:77)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async SubscriptionNotification.parsePrintStats (/home/pi/mooncord/dist/index.js:51779:9)
[info] [2022-04-23T07:44:51.445Z] Agent772#3294 executed command: status

[error] [2022-04-23T07:44:51.719Z] TypeError: Cannot read properties of undefined (reading 'embed_id')
            TypeError: Cannot read properties of undefined (reading 'embed_id')
    at StatusCommand.execute (/home/pi/mooncord/dist/index.js:50641:80)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)

eliteSchwein, 2022-09-26:
Hey there, this is properly fixed in 0.0.6 (which will come out next month)

eliteSchwein, 2022-10-25:
Hey there any updates?

closed 2022-10-25 by Agent772