eliteSchwein/mooncord #286: cannot get invite for my discord bot

andimoto, 2022-08-21:
Hi,

I've installed mooncord on my raspberry running klipper and mainsail.
I don't know how to retrieve moonraker_token or if it is needed at all.

At first I had some errors in moonraker.log saying following
```
Service 'mooncord' not allowed
utils.ServerError: Service 'mooncord' not allowed
```
I've installed mooncord again because I had it in a different directory. After I just moved it to ```/home/pi``` this error disappeared. But still it is very unclear to me how to configure everything, as the wiki is not very detailed for me.

Running ```RESPOND PREFIX=mooncord.invite MSG=get``` in the console I just get ```mooncord.invite``` and nothing else.
Running systemctl gives me following:
```
MoonCord.service - MoonCord
   Loaded: loaded (/etc/systemd/system/MoonCord.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Sun 2022-08-21 10:06:35 BST; 3s ago
  Process: 1594 ExecStart=/usr/bin/npm run start /home/pi/klipper_config/ (code=exited, status=0/SUCCESS)
 Main PID: 1594 (code=exited, status=0/SUCCESS)

Aug 21 10:06:35 mainsailos npm[1594]:     at ConfigHelper.getUserConfig (/home/pi/mooncord/dist/index.js:46648:21)
Aug 21 10:06:35 mainsailos npm[1594]:     at ConfigHelper.loadCache (/home/pi/mooncord/dist/index.js:46641:32)
Aug 21 10:06:35 mainsailos npm[1594]:     at Module.1667 (/home/pi/mooncord/dist/index.js:52622:14)
Aug 21 10:06:35 mainsailos npm[1594]:     at __nccwpck_require__ (/home/pi/mooncord/dist/index.js:53524:43)
Aug 21 10:06:35 mainsailos npm[1594]:     at /home/pi/mooncord/dist/index.js:53693:37
Aug 21 10:06:35 mainsailos npm[1594]:     at Object.<anonymous> (/home/pi/mooncord/dist/index.js:53696:12)
Aug 21 10:06:35 mainsailos npm[1594]:     at Module._compile (node:internal/modules/cjs/loader:1126:14)
Aug 21 10:06:35 mainsailos npm[1594]:     at Object.Module._extensions..js (node:internal/modules/cjs/loader:1180:10)
Aug 21 10:06:35 mainsailos npm[1594]:     at Module.load (node:internal/modules/cjs/loader:1004:32)
Aug 21 10:06:35 mainsailos systemd[1]: MoonCord.service: Succeeded.
```
Also grepping through moonraker.log gives me no useful informations.

Can you help me here?

Best regards
Andi

grantrules, 2022-08-22:
i just installed it and didn't add moonraker_token, just updated "bot_token" from the bot page in the discord dev site.

Once you do that you should be able to do `sudo service MoonCord restart` and see it in `sudo service MoonCord status`

eliteSchwein, 2022-08-22:
Hey there,

moonraker_token is optional and only required when you have forced_login enabled in the Moonraker Config,
about the Bot Token, as @grantrules mentioned, you need to refresh the token on the discord dev site in order to copy it.
And about the Wiki, i will work on a better wiki as soon as the new Update is out.

Greetings from Germany
eliteSCHW31N

andimoto, 2022-08-24:
Hey, 
thanks. Looks like my json file should be ok then. In the status output I get for Active: inactive (dead). If I restart the service and run a status directly after, I get 

```
● MoonCord.service - MoonCord
   Loaded: loaded (/etc/systemd/system/MoonCord.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2022-08-24 18:54:03 BST; 2s ago
 Main PID: 7983 (npm run start /)
    Tasks: 19 (limit: 4915)
   CGroup: /system.slice/MoonCord.service
           ├─7983 npm run start /home/pi/klipper_config/
           ├─7994 sh -c -- node --expose-gc dist/index.js /home/pi/klipper_config/
           └─7995 node --expose-gc dist/index.js /home/pi/klipper_config/

Aug 24 18:54:03 mainsailos systemd[1]: Started MoonCord.
Aug 24 18:54:04 mainsailos npm[7983]: > mooncord@0.0.5 start
Aug 24 18:54:04 mainsailos npm[7983]: > node --expose-gc dist/index.js /home/pi/klipper_config/
```
Then one second later it is inactive again. Is this as supposed?

Also greetings from Germany ;)
Thanks for you help.
I'm also on  discord if this is better for communication...

Andi

eliteSchwein, 2022-09-10:
Hey there,
i didnt had the Time to check Github Issues, sorry about that, are you already on the Community Discord Server? :)

andimoto, 2022-09-11:
Hi,
I also had no time for this. I will get back as far as I will go on with this issue.
Thanks.

eliteSchwein, 2022-10-25:
Hey there any updates?

andimoto, 2022-10-25:
hey, sorry. still had no time. i think i will try it next week.

andimoto, 2022-10-25:
hey eliteSchwein,

thanks for updating the installation part and pushing this issue. I think I managed to install it and looks like mooncord is running. but I still not get any status updates. I pasted your status example into the mooncord.json file. A print is currently running and I should get at least 2 status messages. but the is nothing in my discord channel. I don't know if something is missing.

To be honest, when viewing the usage & config guide, I do not know what to do and what most things are for... The are no detailed informations on the topics I can configure and for example which of them are necessary or just optional. I also had to guess in which file I have to paste the configuration (of course it is the json file because klipper has a completely different syntax)
I just need my printer to send me a status message, and in the future a webcam pic.

Are there any other configs I have to do? When my build is finished, I can send you the content of my mooncord.json file.

Gruß
Andi

eliteSchwein, 2022-10-31:
Hey andi

to enable automatic notifications you need to use /notifyme (for dm notifications) or /editchannel (for channel notifications)

andimoto, 2022-10-31:
Where do I need to configure this? In Discord in the Bots channel?

andimoto, 2022-10-31:
Hey, I just removed MoodCord. Sorry. I installed the update and mooncord service did not run. I don't know why, but I don't want to investigate every time. Maybe I will try it later...

closed 2022-10-31 by andimoto