eliteSchwein/mooncord #305: No respond after closing SSH connection

chopdi, 2022-12-24:
hey when i install mooncord it runs just as long as i keep the ssh connection open. after that it doesn't respond anymore. bot is loaded and installation works without errors.json is also adjusted

eliteSchwein, 2022-12-26:
Hey there, did you type in a message channel on your server where you invited the bot while running the installation? If you didnt do that, this might finish your installation :)

Maxibauser, 2023-11-10:
Hi please check if you`re bot is on the server and can acces the channel!

eliteSchwein, 2023-11-20:
Hey @chopdi i released a update that has a better install wiki (i asked multiple people if its understandable). Please consider a reinstallation :). I will close issues and if the issue remains please reopen this issue.

closed 2023-11-20 by eliteSchwein