eliteSchwein/mooncord #313: Error completing install if using 2FA for Discord

Gil80, 2023-02-20:
Getting this error:
```
An Error occured while registering the command tune
[error] [2023-02-20T23:14:20.731Z] Reason: DiscordjsError: Request to use token, but token was unavailable to the client
```
I believe that this is because I'm using a 2FA token when logging into Discord.

eliteSchwein, 2023-02-21:
Hey there

It looks like you use a wrong token, not sure. I see the error the first time. but it has nothing to do with 2fa, i am also using 2fa and never saw that error.

Greetings from Germany

Gil80, 2023-02-21:
Ok, I don't know what happened but I was able to complete the install.
But then I got this error:
[image]

eliteSchwein, 2023-02-21:
Well you didnt finish the install, you pressed ctrl + c, my guess is that you tried to copy the invite url. On your ssh client when you mark the invite link just press the right mouse button, that should copy the invite url. Ctrl +c kills the installscript and doesnt copy the url

Gil80, 2023-02-21:
Do I need to leave this CLI running in the background?
This is what I see now
[image]

eliteSchwein, 2023-02-22:
no, you invite the bot and write in a channel with your discord account. this will set your permissions and finish the install script (its also mentioned in the "test run")

Gil80, 2023-02-22:
yeah I finally figured it out. Though I read the instructions, it wasn't clear to me what it meant to invite the bot. 

Thanks. All sorted now.

closed 2023-02-22 by Gil80