eliteSchwein/mooncord #318: /editchannel not working - Empty list of channels

ullbergm, 2023-05-29:
I have reinstalled several times just to be sure, clearing the config each time.
I have tried new bot tokens.

When i try to run /editchannel the GUI does not show me a list of channels to choose from:
[image]

Everything else seems to work, no errors on the console and i used the link listed there.
[image]

eliteSchwein, 2023-05-30:
hey there, are you using the command in dm or on a server?

ullbergm, 2023-05-30:
I'm an idiot, lol. I didn't realize it would work in any channel. It's all good.

closed 2023-05-30 by ullbergm