eliteSchwein/mooncord #319: Add Wiki with Discord Usage

theely, 2023-08-20:
Hi there,
unfortunately it's not possible to send a pull request for a wiki page so instead I am opening this issue.

Would be great to have a Wiki Page under the "Usage" section on how to interact with the bot in Discord.
In the Page I suggest to have two sections:

1. Commands: with a list of commands the bot responds to
2. Configuration: with a list of advanced commands to configure to bot (e.g. setting up broadcasting channel)

Thanks

eliteSchwein, 2023-08-28:
Hey there, sorry for the late response.

Yeah i will rework the wiki as soon as the next update is out

eliteSchwein, 2023-11-20:
https://eliteschw31n.gitbook.io/mooncord/usage/discord-commands

Work in progress

eliteSchwein, 2024-08-17:
with 1.0.5 i fully want to migrate to another wiki system that moonraker uses, easier to maintain.

eliteSchwein, 2024-12-08:
done with 1.1.0 on here -> https://wiki.tludwig.dev/mooncord

closed 2024-12-08 by eliteSchwein