eliteSchwein/mooncord #320: Error whilst completing Setup

GitHub-source-42, 2024-01-10:
I get the following error message and set up will not continue
any help would be appreciated

[error] [2024-01-10T19:53:20.150Z] An Error occured while registering the command preheat
[error] [2024-01-10T19:53:20.177Z] Reason: DiscordAPIError: Invalid Form Body
options[1].options[1].name: String value did not match validation regex.
[error] [2024-01-10T19:53:20.179Z] Command Data: {
    "name": "preheat",
    "description": "Preheat a profile or manually.",
    "options": [
        {
            "type": 1,
            "name": "preset",
            "description": "Preheat Preset.",
            "options": [
                {
                    "type": 3,
                    "name": "preset",
                    "description": "Select a Preset.",
                    "options": [],
                    "required": true,
                    "choices": [
                        {
                            "name": "ABS",
                            "value": "ABS"
                        },
                        {
                            "name": "PLA",
                            "value": "PLA"
                        },
                        {
                            "name": "Cooldown",
                            "value": "Cooldown"
                        }
                    ]
                }
            ],
            "choices": []
        },
        {
            "type": 1,
            "name": "manual",
            "description": "Manual Select Temps.",
            "options": [
                {
                    "type": 4,
                    "name": "heater_bed",
                    "description": "Set the Temperature for heater_bed",
                    "options": [],
                    "required": false,
                    "choices": [],
                    "min_value": 0,
                    "max_value": 120
                },
                {
                    "type": 4,
                    "name": "heater_generic heater_bed1",
                    "description": "Set the Temperature for heater_generic heater_bed1",
                    "options": [],
                    "required": false,
                    "choices": [],
                    "min_value": 0,
                    "max_value": 120
                },
                {
                    "type": 4,
                    "name": "extruder",
                    "description": "Set the Temperature for extruder",
                    "options": [],
                    "required": false,
                    "choices": [],
                    "min_value": 0,
                    "max_value": 330
                }
            ],
            "choices": []
        }
    ]
}

eliteSchwein, 2024-01-12:
hey there, it looks like you have 2 heat beds? anyways, you found a interesting bug and i will fix it asap

eliteSchwein, 2024-01-12:
should be fixed with 
369b1a6

GitHub-source-42, 2024-01-14:
Yeah the 2 beds are as there are 2 separate heat zones on my printer
Elegoo Neptune 4 Pro
thanks !
I'll try again now and let you know how I get on

closed 2024-01-14 by GitHub-source-42