eliteSchwein/mooncord #323: /notifyme only works on bot owner, not anyone else

MikevanBreePXL, 2024-03-13:
How to recreate:
- Add bot to server
- Have a friend on a different discord account execute `/notifyme`
- The bot will always give a permission error/hidden message, wether the friend has /admin or not

Expected behaviour:
Anyone in the discord can execute the command, just like /status

MikevanBreePXL, 2024-03-18:
Hi, as stated in the initial post, it doesn't change outcome

eliteSchwein, 2024-03-24:
hey there, sorry for the late response.

currently only discord admins or guild admins can use notifyme, this is more or less a privacy protection.

you can see the config here: https://github.com/eliteSchwein/mooncord/blob/master/scripts/mooncord_full.cfg#L551

MikevanBreePXL, 2024-03-25:
Right on, perhaps this could be a feature request.

closed 2024-03-25 by MikevanBreePXL