eliteSchwein/mooncord #326: Download MoonCord and start the install script - doesn't work

luSSSh, 2024-04-06:
Following the wiki and doing the first step of the install script i get this follow error

admin@klipper:~ $ cd ~/
git clone https://github.com/eliteSchwein/mooncord.git
cd ~/mooncord
bash scripts/install.sh
Cloning into 'mooncord'...
remote: Enumerating objects: 22859, done.
remote: Counting objects: 100% (3900/3900), done.
remote: Compressing objects: 100% (1021/1021), done.
error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8)
error: 4999 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
-bash: cd: /home/admin/mooncord: No such file or directory
bash: scripts/install.sh: No such file or directory

luSSSh, 2024-04-09:
i was able to  download the rest of the packages using this command, and manual running the install.sh

git clone --depth 1 https://github.com/eliteSchwein/mooncord.git

eliteSchwein, 2024-12-08:
thats interesting
i tried multiple times to reproduce this, no issues on my side

closed 2025-05-21 by eliteSchwein