eliteSchwein/mooncord #338: Cannot DM bot for setup

jsinkers, 2024-09-10:
When running the install script, I get to the point where I need to DM the bot. I have invited the bot to the server, it said that it was successfully added, and I can see it listed under integrations with the required permissions. 

However, the bot does not show up as a member of the server, and I cannot find any way to DM it to continue with the setup process, possibly the bot is not online. I'm not very familiar with Discord user so I may have missed something obvious. 

The install script shows:

```
[info] [2024-09-10T04:05:47.940Z] please invite the bot on a Server: https://discord.com/oauth2/authorize?client_id=xxx&permissions=3422944320&scope=bot%20applications.commands

[info] [2024-09-10T04:05:47.943Z] After inviting the bot. Please DM the bot to complete the setup.
[info] [2024-09-10T04:05:47.944Z] personalDiscordUsername
```

Firstly can I confirm whether I should have listed my personal discord tag when running through the install script? 

Any suggestions or recommended steps to troubleshoot would be appreciated

eliteSchwein, 2024-09-11:
hey there, sorry for the late response

thats kinda interesting. maybe a server setting is preventing the bot from joining?

jsinkers, 2024-09-13:
Resolved - doesn't look like it was a mooncord issue, and thanks for the response. There didn't seem to be any discord server settings that needed to change and I was having the same issue on fresh server with default settings. I reran the install script again today (a few days later), and opened the discord invite on a different device/browser and the bot was immediately added to the server as expected. Not sure of the culprit - it may have been a browser extension or browser quirk interfering with discord on my other device.

closed 2024-09-13 by jsinkers